ms-iot/content #160: node.js

nishanthkarthik, 2015-03-16:
node.js 0.12.0 doesnt work on Galileo Gen 1, whereas, 0.11.9 version works fine. I think this should be included specifically on the webpage. This is my first time opening an issue and wasnt sure where to. Thanks.

ooeygui, 2015-03-16:
Thanks for the report. This is a good place to report a problem. If you want to take a crack at getting 0.12.0 working, here's the docs for figuring out what doesn't work: http://ms-iot.github.io/content/Forwarders.htm.

nishanthkarthik, 2015-03-16:
Also, v 0.11.9 doesn't load <b>./ms-iot-wiring.node</b>. I tried 0.10.0.0 and it works with  <b>./ms-iot-wiring.node</b>

ooeygui, 2015-05-03:
Thank you for the report. I'll look into this.

ooeygui, 2015-05-11:
Node.js is supported on Windows 10. We won't be doing new feature work on Galileo. See our FAQ on this - http://ms-iot.github.io/content/Faqs.htm#galileo

closed 2015-05-11 by ooeygui