ms-iot/content #222: Instructions to FlashSD card if you don't have a physical computer

GITGW, 2015-05-03:
This is all good and fine, but most of us don't have a physical computer running Windows 10, so as the site mentions VMs don't have access to physical drives or the card reader, so you can't format and partition a physical disk (you can virtualize it and install it onto the virtual drive, create similar partitions on the physical drive and copy the contents of the virtual drive onto it, and copy the BCD of the virtual hard drive, but that's not 7what we want to do). To be able to flash the Windows Core image onto the SD card we need the latest version of a tool called dism.exe. The latest version of this tool is included with Windows 10 (that's why this manual says that you need Win 10 to do it), but it's also available as part of the Windows Assessment and Deployment Kit and luckliy enough Microsoft has made it available for everyone, so we don't need to scrap it with its dependant DLLs from Win 10.

To be able to flash the image from a computer that's running Windows 7 or higher (I haven't tried from anything below it) we need to first download and install Windows Assessment and Deployment Kit (ADK)for Windows 10.
 Download page: https://msdn.microsoft.com/en-gb/win.../dn913721.aspx
 Direct download: http://go.microsoft.com/fwlink/p/?LinkId=526740

JeremyBYU, 2015-05-03:
I did this and flashed successfully.  However when put the micro sd card in the Raspberry PI 2 it does not boot up (just red and green light come on, no blinking)..... Any help would be appreciated.

Note about the sd card, its Samsung  UHS-I card 32 gb.  Its showing up as 60~ MB in windows explorer after flashing, diskpart lists it as 29 gb and 7gb used though... Not sure if any of that matters.

Thanks,

GITGW, 2015-05-04:
Hi Jeremy

First boot takes a while. I successfully booted . I think it took 10 minutes and one restart. 

You didn't say if you had a monitor keyboard and mouse attached, 

What version of OS are you running?

I think the issue is outside of using Dism from the Windows 10 ADK.

ooeygui, 2015-11-07:
We are considering alternatives, but at this time Windows 10 is the supported solution.

closed 2015-11-07 by ooeygui