ms-iot/content #264: Node WinRT package

SidneyAndrewsOpsgility, 2015-05-12:
While working on the sample "Blinky Node.js Server Sample (Windows Universal)", I noticed that it references a "winrt" node package.

```javascript
var winrt = require("winrt");
```

If you follow the instructions exactly on a machine with a clean Windows 10 + VS2015 RC image, the application will fail to debug because the node application cannot find the WinRT module

> 0x800a139e - JavaScript runtime error: Cannot find module 'winrt'

Where can I find or reference the 'winrt' node module?

munyirik, 2015-05-13:
Hello,
That's an error in the doc. Use var uwp = require("uwp"); instead. Thanks for raising the issue, it will be fixed soon.

munyirik, 2015-05-13:
The documentation has been fixed.

closed 2015-05-19 by ooeygui