ms-iot/content #389: Broken download links

Impulss, 2015-06-12:
Windows 10 IoT Core for Raspberry Pi 2
Windows 10 IoT Core for MinnowBoard MAX
NTVS IoT Extension Beta 2015
PTVS for IoT Preview 2015

All 4 are broken. The rest are good.

sungow, 2015-06-12:
I tried windowsondevices.com and clicked "Download" link and was able to get to https://connect.microsoft.com/windowsembeddedIoT/Downloads/DownloadDetails.aspx?DownloadID=57782 which is our download link. Can you explain under which scenario you are seeing this issue of broken links.

ooeygui, 2015-06-17:
It looks like you have not specifically signed for the program. You'll need to use this link [https://connect.microsoft.com/windowsembeddediot/SelfNomination.aspx?ProgramID=8558](https://connect.microsoft.com/windowsembeddediot/SelfNomination.aspx?ProgramID=8558)

From the [getting started guide](http://ms-iot.github.io/content/en-US/win10/SetupRPI.htm)

Please sign up on Microsoft Connect, which is our software release management platform.
* You will need to have a Microsoft Account
* If you are already signed up with our program on [Microsoft Connect](https://connect.microsoft.com/windowsembeddediot/SelfNomination.aspx?ProgramID=8558), (or are not sure if you signed up earlier), you will just see a _blank page or broken links_.
* If you have not signed up for our program on Connect before, it will prompt you to create a social profile and accept license agreements. Follow the step-by-step instructions for signing up for Microsoft Connect to get going.

closed 2015-11-04 by ooeygui