ms-iot/content #551: Errors in the MemoryStatus Node.js (Console Application) Sample for MBM

NNM11, 2015-08-14:
- Host PC running Windows 10 Ent (10240)
- MBM running Windows 10 IoT Core (10240)

[Copy Node.js to MBM/Pi2]
1. Open up an explorer window on your PC and enter \\<IP address of your device>\C$  - enter credential 
2.  Not clear where to run this command from?   "[Run & 'C:\Program Files (x86)\Node.js (chakra)\CopyNodeChakra.ps1' -arch <ARM | x86 | x64 > -ip <Device IP Address>. "]
* Need to open a PowerShell window, then run the command
3. ERROR - in accessing the node_module 
-----------------------------
Copyright (C) 2015 Microsoft Corporation. All rights reserved.

PS C:\WINDOWS\system32> & 'C:\Program Files (x86)\Node.js (chakra)\CopyNodeChakra.ps1' -arch x86 -ip 192.168.0.113
Copying C:\Program Files (x86)\Node.js (chakra)\ to \\192.168.0.113\c$\Node.js (Chakra)\ ...
Copy-Item : Access is denied
At C:\Program Files (x86)\Node.js (chakra)\CopyNodeChakra.ps1:38 char:5
+     Copy-Item -Destination {Join-Path $destinationPath $_.FullName.Su ...
+     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : NotSpecified: (:) [Copy-Item], UnauthorizedAccessException
    + FullyQualifiedErrorId : System.UnauthorizedAccessException,Microsoft.PowerShell.Commands.CopyItemCommand

Copy completed

Copying https://github.com/ms-iot/ntvsiot/releases/download/0.12.6.2-exe-x86/node.exe to \\192.168.0.113\c$\Node.js (Chakra)\ ...

ComputerName     :
RunspaceId       : 4b8e67e0-905f-48c3-98f9-16957afd3271
EventIdentifier  : 2
Sender           : System.Net.WebClient
SourceEventArgs  : System.ComponentModel.AsyncCompletedEventArgs
SourceArgs       : {System.Net.WebClient, System.ComponentModel.AsyncCompletedEventArgs}
SourceIdentifier : Finished
TimeGenerated    : 8/14/2015 2:56:00 PM
MessageData      :

Copy completed
PS C:\WINDOWS\system32> 
----------------------------
* Workaround: copy from Windows Explorer the "Node.js (chakra)" to the IoT Core device C$

4. ERRROR - Build the MemoryStatusAddOn.vcxprj.   
LINK : fatal error LNK1181: cannot open input file 'node.lib' [D:\Archive-Legacy\NodeAddOn\build\MemoryStatusAddon.vcxproj]
* The project link path reference to "C:\Program Files (x86)\Node.js (chakra)\sdk\x86"  where the actual folder is "C:\Program Files (x86)\Node.js (chakra)\sdk\ia32".  
5. Load the project file into VS2015, fix the link path and build success.

munyirik, 2015-11-11:
Hi @NNM11, the issues will be resolved in the next release. For the first issue with the file copy, the workaround is to:
1. Copy the program files\node.js (chakra) folder manually to your device.
2. Then copy https://github.com/ms-iot/ntvsiot/releases/download/0.12.6.2-exe-<replace arch>/node.exe to the folder to replace the node.exe (since you're using x86 you don't need this step).

IoTGirl, 2016-05-31:
Fixed

closed 2016-05-31 by IoTGirl