ms-iot/content #554: Windows_10_IoT_Core_Packages.msi

reneherrero, 2015-08-16:
Hi,

There´s a broken link in this page:
https://ms-iot.github.io/content/en-US/win10/ImageCreation.htm

Under the section "Install Windows 10 IoT Core packages", the link "ISO" has the value "todo: add installation location for ISO" which obviously doesn´t resolve.

That ISO is suppose to contain Windows_10_IoT_Core_Packages.msi.

Anyone know where I can find it?

Thanks,

René

cmaloney-msft, 2015-08-26:
Hi René, 

We are sorry but the OS packages ISO/MSI are only currently available directly via Microsoft support please contact your Microsoft representative if you are interested in signing up for access.

We are in the process of updating the ImageCreation.htm page to reflect this. Thanks for pointing it out. 

Christopher Maloney [Microsoft]

reneherrero, 2015-08-27:
Hi Christopher,

Thanks for your reply.

I'm a little confused.  I was under the impression all this IoT stuff was going to be free?  Or does it just apply to the pre-built Minnowboard Max/Raspberry PI images?

I took a look at my MSDN subscription and I can see Minnowboard Max and Raspberry PI ISOs, but that's it.

And, while I'm here, any chance the BeagleBone Black will be supported any time soon?  Forgive me for trying...  sweet little board.  :)

Kind regards,

René

mburumaxwell, 2015-10-19:
Hi Chistopher,

I have two questions
1) Does the requirements to contact Microsoft support mean there is at least some purchase fees?
2) What happens to some independent developer who does not want to use a raspberry pi or minnow board max but would like to run Windows IoT Core on something like a Cubeboard3?

Thank you in advance :-)

Wack0, 2015-10-28:
about Windows_10_IoT_Core_Packages.msi, I have it, but it's of an old build.

It can be used with tools inside WP Adaptation Kit or Win10 WDK to build custom FFUs.

jryfar, 2015-11-16:
Hi René,

We are working to make the generic packages that Wack0 noted available in the very near future (weeks) through regular distribution channels.

We also will have commercialization terms available at release time that will make the licensing very clear.  I apologize that I can't share more information early, but hopefully the indication that it is coming soon will give you some solace. 

Thanks for your interest!  We are very anxious to get this stuff out to you.

-Jason

marcusrussell, 2015-11-16:
Jryfar's comment on 11/15 resolves #554

marcusrussell, 2015-11-16:
Closing issue per previous comment.

closed 2015-11-16 by marcusrussell