ms-iot/content #573: Windows 10 IoT Core WIFI Adapters

Injac, 2015-08-20:
Hi,

this article needs to be updated, with more in depth information, like a video for each adapter showing it working and with the firmware versions (adapter firmware, version) used.

https://github.com/ms-iot/content/blob/develop/en-US/win10/SetupWiFi.md

I bought the TP-LINK TL_WN725N for my MBM and I cannot make it work.

cmaloney-msft, 2015-08-20:
Hi Injac, can you share what issues you are seeing with the TL_WN725N? Does the UI show as you having a WiFi adapter install?

Injac, 2015-10-09:
Hi,
Sorry for the late reply. No the UI did not show up. The adapter has been defective. Everything works fine now.

ooeygui, 2015-11-07:
We now have a hardware compatibility tab on windowsondevices.com - http://ms-iot.github.io/content/en-US/win10/SupportedInterfaces.htm

closed 2015-11-07 by ooeygui