ms-iot/content #591: Failed to commit pending deployment

ehigh2014, 2015-09-02:
I use Visual Studio 2015 develop an app and debug on the raspberry pi 2 success.
But when I get the package and use the appx manager from the web to commit deployment, I failed.
Error code : 0x80004005, could you help solve this problem?

ehigh2014, 2015-09-02:
Reference : http://ms-iot.github.io/content/en-US/win10/tools/Webb.htm

bfjelds, 2015-11-10:
What kind of app are you using (Blank Application, Background Application, etc)?  Language (C++, C#, etc)?  Do you have the same problem using Blinky or HelloWorld from https://github.com/ms-iot/samples?

IoTGirl, 2016-05-31:
Closing this issue as there has been no response from the original poster and AppX documentation is available at https://developer.microsoft.com/en-us/windows/iot/win10/backgroundapplications

closed 2016-05-31 by IoTGirl