ms-iot/content #606: Raw images for RPi2

lukeadams, 2015-09-08:
Hey guys, still setting up my RPi for W10. I think it would be great if you provided a writeable iso (for those of us on OSX or Linux) to quickly get started. I have Windows in a VM for the tooling, but it would help me get started just that much faster.
Also helpful would be to make the current download a zip instead of an ISO. It doesn't really make sense to me to encapsulate an installer that way...

HeisenbergUncertain, 2015-09-09:
I too have some good stuff going for us. gonna get all the good acronyms that are the ones with all the bad letters in them - who knows what they could stand for......
well, it's close enough pay for government vork - fork, sorry !!!!!!!!!!

[image]

lukeadams, 2015-09-09:
Ech. Sorry about the wordy post – I was exhausted at the time.
I updated it for clarity (the acronyms aren't meaningless; the people who could resolve this would know them, otherwise you are a quick google search away...)

SunboX, 2015-10-03:
Jep, a simple *.img file would be a lot easier to install. Would appreciate it.

marcusrussell, 2016-01-14:
Thanks for this feedback, but at this team we do not plan to package releases as .img, and instead will make releases via .iso packages.

closed 2016-01-14 by marcusrussell