ms-iot/content #640: Blinky WebServer Sample Can't work

Myfreedom614, 2015-09-30:
Follow this document: http://ms-iot.github.io/content/en-US/win10/samples/BlinkyWebServer.htm
>For our sample app, we have hosted a simple client in our WebServer on the Windows Windows IoT Core device. It can be accessed via the IP address of the board and the port (something like http://123.456.789.0:8000 where you replace 123.456.789.0 with the IP address of the device the server has been deployed to). The webserver client can then be used to toggle the LED on and off (see screenshot below).

On my side, **http://<IP/machinename>:8000/Blinky.html?state=on** can't trigger the backgroundtask event

ooeygui, 2015-11-07:
Assigning to @bfjelds; perhaps the background task hasn't been deployed?

rachitb777, 2015-11-10:
Hi Myfreedom614,
Did you deploy both the applications? ie. Blinky app and the WebServer app ? Note that Blink app is the one that you should launch. It launches the webserver app.

MichaelBross, 2015-12-21:
I had the same problem after first deploying the WebServer app then running the BlinkyWebService app, as instructed. 

I tracked the problem down to an apparent change in the WebServer app's Package family name. Specifically BlinkyWebService attempts to connect to appServiceConnection.PackageFamilyName = "WebServer_hz258y3tkez3a" but the WebServer app's package family name (shown in the Package.appxmanifest in Visual Studio on my machine) is WebServer_1w720vyc4ccym. 

Just change the BlinkyWebService MainPage.xaml.cs line 47 to read appServiceConnection.PackageFamilyName = "WebServer_1w720vyc4ccym"; or whatever Package family name appears in WebServer Package.appxmanifest in Visual Studio. The Package family name does not appear in the WebServer Package.appxmanifest source file found here on GitHub. At least I couldn't find it. So I suspect the name may be automatically generated and may be different for every user. Reinforcing this belief is the fact that the Package family name appearing in Visual Studio is read only (grayed out).

edwyn, 2016-01-03:
@MichaelBross thanks for you post, changing the appServiceConnection.PackageFamilyName did the trick for me, for me in the grayed out area in the Package.appxmanifest also showed WebServer_1w720vyc4ccym ..