ms-iot/content #644: Please add support for eGalax Touchscreen Controllers

SunboX, 2015-10-03:
Many Raspberry Pi Touchscreens are using a eGalax Touchscreen Controller (http://www.eeti.com.tw/) like this one I'm owning:
http://www.eleduino.com/5-Inch-480-*800-HDMI-Input-Touch-Screen-for-Raspberry-PI-2-B-B-and-Banana-Pro-Pi-Beaglebone-Black-p10440.html

By now there's no way to get the touch interface to work with Windows 10 IoT Core. Please add support for these controllers. many will benefit from it.
https://www.bing.com/search?q=eGalax+Raspberry+Pi

marcusrussell, 2015-11-25:
Hello SunboX,

Thanks for your feedback.  IoT Core compatibility with a wide range of devices and controllers is a priority for us, and we're currently investigating  support for this device and others like it.  We'll update the community with our plans as soon as our evaluation is complete. Stay tuned.

IoTGirl, 2016-06-01:
Hello Sunbox,
Your request has been recorded but we do recommend that you reach out to the OEM of your hardware and ask them for IoT Core Drivers to make sure they are aware of your interest in using their hardware with IoTCore. 
Supported hardware will be added to the Hardware Compatability List at https://developer.microsoft.com/en-us/windows/iot/win10/SupportedInterfaces.htm. 
Sincere thanks for your input,
IoTGirl

closed 2016-06-01 by IoTGirl