ms-iot/content #645: Please add support for Edimax WiFi USB dongles

SunboX, 2015-10-03:
Many Raspberry Pi projects are using a Edimax WiFi USB dongle like this one:
http://www.edimax.com/edimax/merchandise/merchandise_detail/data/edimax/global/wireless_adapters_n150/ew-7811un

By now Windows 10 IoT Core only supports the official and very rare Raspberry Pi USB WiFi dongle with a Broadcom chip set. Sadly, there's no other dongle with this chip set available.

So actually there's no way to get these widely used Edimax dongles to work with Windows 10 IoT Core (even if Edimax states they will work with Windows 10!). Please add support for these WiFi dongles. many will benefit from it.
https://www.bing.com/search?q=Edimax+Raspberry+Pi

ravindrank, 2015-10-19:
+1
The Edimax adapter mentioned is a RTL8188CUS (which is same chipset used in TP-LINK WN725N  already supported for Raspberry pi 2 as per release notes of 10556) .

Also, why I don't see any drivers for the TP-LINK WN725N in c:\windows\inf?

ricl, 2015-10-19:
http://www.ebay.co.uk/itm/like/361362425807?adgroupid=13585920426&hlpht=true&hlpv=2&rlsatarget=kwd-75951457866&adtype=pla&ff3=1&lpid=122&poi=1006976&ul_noapp=true&limghlpsr=true&device=c&chn=ps&campaignid=207297426&crdt=0&ff12=67&ff11=ICEP3.0.0-L&ff14=122&viphx=1&ops=true&ff13=80

also works out of the box (same Broadcom chip as the official adapter).

ravindrank, 2015-10-20:
Thanks ricl. Aware of that.. Most of those ship from UK .. to get in US.. the cost just doubles up and adding delay.. 
Point here is .. How can get a wider range of hardware to work on Windows 10 IOT on Raspi2...

marcusrussell, 2015-11-10:
Hello Ravindrank,

We've heard your feedback, and we're working on extending WiFi dongle support in upcoming releases.  We will update supported peripheral documentation along with the public release of new IoT Core versions.  You can find up-to-date device compatibility lists at the following link: http://ms-iot.github.io/content/en-US/win10/SupportedInterfaces.htm.

ravindrank, 2015-11-11:
Thank you Marcus.

SunboX, 2015-11-11:
sound's good, thank you! :+1:

closed 2015-11-17 by marcusrussell

lappiesja, 2016-02-27:
Seen this elsewhere at this link, just scroll down:
https://social.msdn.microsoft.com/Forums/en-US/05fe7028-3d1b-454e-a4e8-f614374390a1/installing-usb-wireless-edimax-driver?forum=WindowsIoT

Just compiling the steps others have listed.

1. Download the driver zip from http://www.edimax.com/edimax/download/download/data/edimax/global/download/for_home/wireless_adapters/wireless_adapters_n150/ew-7811un

2. Put SD card in PC and copy the 3 files that were in the zip to <sdcard>:\temp. Then eject.

3. Put SD card in the Pi and boot.

4. Connect with Powershell (detailed in https://ms-iot.github.io/content/en-US/win10/samples/PowerShell.htm)

5. cd to c:\efisp\temp

6. Run the command below.

devcon install netrtwlanu.inf "USB\VID_7392&PID7811"

I got a "RemoteException" but it worked anyway. 

7. exit powershell session

8. Restart Pi

9. Go to http://minwinpc:8080/wifimanager.htm to configure.

This worked for me. Thanks for everyone who did the hard part.