ms-iot/content #676: Release notes incorrectly mentions support for TPLink WiFi dongle on Raspberry Pi2

ravindrank, 2015-10-19:
Release notes indicates:
Support for the TPLink WiFi dongle on Raspberry Pi2

We are unable to find any driver in C:\Windows\Inf on RPi2 and do not believe that the wifi adapter is supported on RPi2 (only minnowboard)

ricl, 2015-10-19:
http://www.ebay.co.uk/itm/like/361362425807?adgroupid=13585920426&hlpht=true&hlpv=2&rlsatarget=kwd-75951457866&adtype=pla&ff3=1&lpid=122&poi=1006976&ul_noapp=true&limghlpsr=true&device=c&chn=ps&campaignid=207297426&crdt=0&ff12=67&ff11=ICEP3.0.0-L&ff14=122&viphx=1&ops=true&ff13=80

also works out of the box (same Broadcom chip as the official adapter).

ravindrank, 2015-10-20:
Thanks ricl. Aware of that.. Most of those ship from UK .. But anyway the concern is different here.

cmaloney-msft, 2015-10-20:
Thanks for catching this. You are correct this build does not support the TPLink dongle on RPi2 and there was an error in the release notes. We are updating the release notes.

andreabeggiato-geekcups, 2015-10-20:
Amazon deliver me this dongle today. There are some plans to support this type of dongle?

cmaloney-msft, 2015-10-20:
Yes, we are working on this for a future release.

ravindrank, 2015-10-23:
Thanks for your ack :)

volkansarioglu, 2015-10-31:
Are you kidding me ? 
We are spending money and time for it, how do you make a mistake in supported device list?
Im not use Windows IoT anymore cause; 
 if you make a mistake in supported device list, you can make a mistake in every where.

andreabeggiato-geekcups, 2015-11-06:
Hi guys,
you have remove the wrong release note from here http://ms-iot.github.io/content/en-US/win10/ReleaseNotes.htm ,but you have add RPI2 compatibility in the hardware section here http://ms-iot.github.io/content/en-US/win10/SupportedInterfaces.htm

HerrickSpencer, 2015-11-06:
Thanks for the notice, Andrea.
This is currently in [PR to be removed](https://github.com/ms-iot/content/pull/690)

ooeygui, 2015-11-07:
This has been removed from the hardware guide.

closed 2015-11-07 by ooeygui

byGao, 2015-11-15:
hi cmaloney-msft:
the following page still have wrong list..(about wifi dongles tag..with TP-WN725N part)
http://ms-iot.github.io/content/en-US/win10/SupportedInterfaces.htm

and... when will support TP-WN725N?
please give me a page to follow.
thank you

judavi, 2015-11-16:
Yes, it's still there... 
[image]

ricl, 2015-11-16:
It has come (and gone) from that page a few times. :-)

Have patience, it is reported that it IS supported in the, yet to be fully
released - only leaked, 10586 IoT release.

byGao, 2015-11-17:
ok...
Thanks for your reply.

maksii, 2016-03-09:
Ahh, amz. It's still in list and it's not working. Are you going to fix it?