ms-iot/content #681: IoTBrowser App dies after less than a day

realwarder, 2015-10-28:
Running the sample app IoTbrowser, we are running it pointing to a web site that updates a live data feed.

The browser app dies within 24 hours.  Various errors are the cause, from EM_WATCHDOG_TIME terminating the app process to a fault in EdgeHTML.dll.

For a simple terminal display, being able to run an app permanently is important.

In the interim, is there a method for an app that terminates or crashes to auto-restart?

Thanks

ooeygui, 2015-11-07:
GitHub triage - @DavidShoe Could you follow up with this?

realwarder, 2015-11-07:
After spending a few days running the browser on various sites I think it's likely a javascript memory leak bug in the website we were monitoring eventually ending up causing this crash of the IoTBrowser app, not inherently the app.

However the request for how can we setup the platform to automatically restart the default app on a crash is important.

Thanks

DavidShoe, 2015-11-13:
We have seen your request and I am doing some research now.  We will get back to you shortly.

DavidShoe, 2015-11-20:
If you have used the web interface and set the iot browser application as the default application.  When the device starts up it should boot into the browser application.  In this state if the browser crashes for any reason the OS should restart it automatically.  Is this not happening for you?  

One note: the OS has a counter on the number of times this happens. After 10 crash / restart cycles the device will "blue screen", and reboot.  This will reset the counter and the cycle will continue.

ooeygui, 2015-11-20:
@realwarder Please provide us with a sample application (or the feed) which reproduces the crash you are seeing, we will see what's happening.

closed 2015-11-20 by ooeygui