ms-iot/content #751: During Serial port communication we are getting PortName as readonly

sheethalkumarsm, 2015-12-18:
Hi

During Serial port communication we are getting Port Name as read only we used  CP2102 USB to UART Bridge Controller and RS 232. when we debug the code we found that port name is showing as blank.to update the port name it is showing as read only as below
"Property of indexer "SerialDevice.Portname cannot assigned to  -- it is readonly"
meantime we tried to update the "COM1" from serialDevice.getselector("COM1"), nothing is coming. 
Please suggest us how to update Port name

Regards
Sheethal

sheethalkumarsm, 2015-12-18:
We used Rapberian pi 2, Windows IOT

IoTGirl, 2016-06-01:
Hi sheethalkumarsm,
There was a lengthy discussion on USB-Serial dongles on MSDN at https://social.msdn.microsoft.com/Forums/en-US/682600bd-c0d3-4c12-9760-4b2f1c539f83/how-to-enable-a-usbtoserial-converter-on-raspberry-pi-2?forum=WindowsIoT.  

At this time the dongle you mention can be found in the HCL at https://developer.microsoft.com/en-us/windows/iot/win10/supportedinterfaces which includes a link to the sample project that has been tested with this hardware.

If you have further issues, please provide more details as to what build you are targeting and more details around what else you may also have connected.

Sincerely,
IoTGirl

closed 2016-06-01 by IoTGirl