ms-iot/content #755: Where can I get OS Kits and BSP Packages?

AceHack, 2015-12-27:
I can't find out how to get any of this.  Whenever I see that I see people say contact your MS rep, I don't have a MS rep how do I get one and who do I get one to contact about this?

SocalSam, 2015-12-29:
Hi AceHack!
If you follow the instructions you will get the Python stuff required.  But you don't need the BSP package, just the Windows 10 IOT Core, which can be found at: https://dev.windows.com/en-US/iot, then navigate to the download button for the Raspberry Pi 2+.
Hope that that helps

AceHack, 2015-12-30:
I'm running fine on Raspberry Pi, Minnow, and Dragon Board, the problem is these all use images customized and created by Microsoft.  I want to use things like Windows ICD, trying to follow this tutorial for one.  https://msdn.microsoft.com/en-us/library/windows/hardware/dn916104(v=vs.85).aspx  I would like to explore many other things as well but they start with the OS Kits and BSP kits.  Thanks.

IoTGirl, 2016-02-24:
See the blog post at https://blogs.msdn.microsoft.com/iot/2015/12/14/windows-10-iot-core-image-creation/ for full details.

AceHack, 2016-02-25:
Thanks, that's exactly what I was looking for.

closed 2016-05-31 by IoTGirl