ms-iot/content #780: TP-LINK TL_WN725N - RPI2 - Disconnects

ramonmata, 2016-01-26:
Compatible hardware list shows TP-LINK TL_WN725N as OK for Raspberry Pi 2, I just got a new one and it works but after a few minutes it gets disconnected  from network.

http://ms-iot.github.io/content/en-US/win10/SupportedInterfaces.htm

So I plug an ethernet cable to the RPI2, and went to Device Portal, usb device still there, but disconneted, and won't list available networks

* Realtek RTL8188EU Wireless LAN 802.11n USB 2.0 Network Adapter
* Device Model: Raspberry Pi 2 Model B
* OS Version: 10.0.10586.63

Found some other Issues #686 #690 but it is unclear to me if it is supported or not

maksii, 2016-03-09:
It's even not listed as usb device in my case. Not works on stable or insider builds.

maksii, 2016-03-27:
Any help on this? or at least advice which way to go?

HerrickSpencerMSFT, 2016-03-28:
Maksii, Sorry you are having an issue with this. While I don't have this hardware to test for you, I checked with internal testers, and it seems to have been previously reported as supported in 10586 IoT release (later than ramonmata's issue here).  Can you tell me which build you are working with? we have Insider build of 14262 out now as well. I will see if I can get someone to test on your build if you are on one of these two.
Thanks!

maksii, 2016-04-06:
@HerrickSpencerMSFT Thanks for your reply. Could you please specify which version of TL_WN725N should be currently supported? Looks like I have v1 and maybe it will works with only v2?

> Can you tell me which build you are working with?
> _Not works on stable or insider builds._

I have issue on 10586(same as @ramonmata )  and 14279(latest insider).

IoTGirl, 2016-06-08:
Hi Ramonmata & Maksii,
10586.218 & 14342 are the latest builds. Could you please confirm you are still getting the same behavior?
Sincere thanks!
IoTGirl

maksii, 2016-06-11:
@IoTGirl 
Same problem:

Device Model: Raspberry Pi 2 Model B
OS Version: 10.0.10586 - (latest from iot dashbord)
Device Model: Raspberry Pi 2 Model B
OS Version: 10.0.14342.1000 - (latest from download portal)

https://onedrive.live.com/redir?resid=59C054BCBDAF56D7!157063&authkey=!AMH7zulvlhx4D3I&ithint=folder%2cpng

jordanrh1, 2016-09-20:
It looks like there is no driver for `USB\VID_0BDA&PID_8176` in build 10586, which would explain why you're seeing error 28.

This device looks to be supported by `netrtwlanu.inf` in the Anniversary Update (build 14393) or higher. 

I have a TP-LINK TL-WN725N, but it reports a hardware ID of `USB\VID_0BDA&PID_8179`. This device seems to be working stable on build 14393.

If this issue occurs on build 14393 or higher, please reactivate the issue, and please include the output from the following comamnds:

    devcon status usb\*
    devcon stack usb\*
    Reg query "HKEY_LOCAL_MACHINE\SOFTWARE\Microsoft\Windows NT\CurrentVersion" /v BuildLabEx

jordanrh1, 2016-09-20:
I have a TP-Link TL-WN7215N and it reports a hardware ID of `USB\VID_0BDA&PID_8179`. This device is supported by netrtwlanu.inf on build 14393 or higher (Anniversary Update).

The model that you have (`USB\VID_0BDA&PID_8176`) is not supported.

The device has been connected for over an hour and is successfully pinging the web. If you experience problems with `USB\VID_0BDA&PID_8179` on build 14393 or higher, please reopen the issue and include output from the following commands

    ver
    devcon status usb\*
    devcon stack usb\*

jordanrh1, 2016-09-21:
@ramonmata 
 - What is the hardware ID of your device (`devcon status usb\*`) ?
 - What is the make and model of the WiFi access point?
 - How is the access point connected to the internet?
 - Is the device inside an enclosure or near metal?
 - Are there any other USB devices plugged in?
 - What is the output of `netsh wlan show all` ?

zeeshanfurqan, 2016-10-31:
@ramonmata 
Hi there,
Can you please provide answers to above questions posted by Jordanrh1?
Thanks!

zeeshanfurqan, 2016-12-02:
@jordanrh1 
Since we don't have info from the user, should we close this issue?

closed 2016-12-02 by zeeshanfurqan

kikeledesma, 2016-12-31:
Hi, 
I have the some problem with Arduino connect to Raspberry Pi 3 via USB.
The Arduino has the Firmata and comunicate ok with laptop by USB ports.

The drive in Windows IoT (OS Version: 10.0.14393.0) on Raspberry is:
USB\VID_1A86&PID_7523\5&3753427A&0&4
    Name: USB2.0-Serial - StatusCode : 25191424 - ProblemCode : 28.

The RemoteBlinky app no display the Arduino in the USB list.

Best Regards,