ms-iot/content #790: WPA2 Enterprise on TP-LINK TL_WN725N  RPI2

timothystewart6, 2016-02-06:
When I try to connect to a WPA2 Enterprise WiFi connection from the web portal it tells me to use the client to connect.  When I use the client it does not allow me to enter a username and password, only a  password.

IoTGirl, 2016-03-07:
Hi Timothy,
Can you clarify if this is a home or work network and specific WPA2 authentication method is being used?  For example:  “WPA2-PSK, WPA2-PEAP, WPA2-PEAPv0/EAP-MSCHAPv2…”  (there are quite a few variations… https://en.wikipedia.org/wiki/Extensible_Authentication_Protocol#IEEE_802.1X). 
Sincere thanks,
IoTGirl

.

mittelman, 2016-03-17:
I'm having the same problem as well with the Dragonboard 410c. I'm a university student and our wireless networks use WPA2-Enterprise for encryption and authentication. Specifically - WPA2-PEAP-MSCHAP2. I can see all the available networks in the Windows 10 IoT settings, however when trying to connect to a secure network, I'm being asked only for the password and not the username. And, of course, connection to these networks immediately fails.

IoTGirl, 2016-06-03:
Answered on MSDN: https://social.msdn.microsoft.com/Forums/en-US/f8ea9432-ac37-45fd-97a2-63e6ebc56997/how-to-connect-to-university-wifi-requires-username-and-password?forum=WindowsIoT

_We actually ship netcmd.exe with our images and we use the same tool to connect to a Test Wifi SSID at Microsoft which uses a username/password as well. So the thought is that this tool could potentially unblock the customer. Prerequisites, the device need to be connected via wired network first.

• After connecting to the device with wired network, open PowerShell
•Ensure the SSID is accessible in that location
•In PowerShell, run this command:_

> netcmd /ActionType:ConnectName /Enterprise:true /Ssid:<ssid_name> /UserName:<user_name> /Domain:<domain_optional> /Password:<password>

closed 2016-06-03 by IoTGirl