ms-iot/content #804: Windows 10 IoT Dashboard

gravityFlower, 2016-02-13:
Problems:
1. Downloads list of devices (seemingly) every time
Every time the program is executed, it seems to load the available device types from the web and therefore (depending on your current traffic and network speed) is displaying an empty device list while it downloads these data. At my laptop this causes an delay of about 5 Seconds before the list of devices is displayed. 

2. Unable to select an existing "iso"
Upon selection of the device, i am only able to download the iso, not select it if i already downloaded it by accident. I had to track the file with Handles, abort the download and replace the iso with the according filename.

2.a Did you ever check how the program act/reacts when there is no internet connection?
I disabled the network adapter (just for fun). It immediately!!! displayed the device types, but still there was only the "Download and install"-Button. On clicking that button, it used the existing iso. But on selecting another device (for which there was no image downloaded) it just displayed the erro that it couldn't resolve the remote name.

3. The "iso"
Why download an iso if it is an msi???

4. "I agree to the software license conditions" 
Every time i click on this, it opens the browser and loads the page.

5. Link to software license conditions is broken
go.microsoft.com/fwlink/?LinkID=70396 -> "We are sorry, the page you requested cannot be found."

6. Program does not check if it can actually write to the storage device
I've written (at least i thought) the image to disk and started my RPi2, but was surprised to see that the previous image was still there. I checked the SD-Card adapter and  and saw that the write lock was activated. I only saw the glimpse of the dism program but did not receive a warning or error message.

Expected fixes:

1. Immediately display the available list of devices and add a button after the drop-down list saying "Check for updated images or devices". And don't try to download the list every time. What if i use a mobile network adapter? Use an Expiry date (why download it when it was downloaded one minute ago).

2. Allow importing existing isos (you can always try to check the sha1/sha256) instead of only allowing "iso" download.

2a. Check network availability (warn if only mobile network adapter are available)

3. Don't diguise the installer in an iso? Why conceal it anyway?

4. Only the word software license agreement should be the link, not the whole sentence (would be nice to check the checkbox on clicking on the text  before the link).

5. Fix the link!

6. Check the return status of the dism.exe?

IoTGirl, 2016-06-02:
Hi Gravity flower,
I am reviewing your issues above and have logged them for review by the IoT Team.
Sincerely,
IoTGirl

IoTGirl, 2016-07-05:
Hi Gravity Flower,
This issue has been marked as fixed and closed by the IoT Core Team.  Please review the latest drop.
Sincerely,
ioTGirl

closed 2016-07-05 by IoTGirl