ms-iot/content #850: BLEGatt not possible in insider preview due to missing file

mjmeans, 2016-04-02:
The instructions at https://ms-iot.github.io/content/en-US/win10/samples/BLEGatt1.htm fail with the latest several versions of insider preview because C:\Windows\System32\IoTBluetoothPairing.exe is missing from the insider preview builds.

mjmeans, 2016-05-19:
Another month gone. I just installed 10.0.14342.1000. Sorry to say that this is still not resolved. IoTBluetoothPairing.exe still does not exist as indicated in the https://ms-iot.github.io/content/en-US/win10/samples/BLEGatt1.htm instructions.

IoTGirl, 2016-06-03:
Hi mjmeans,
Have you tried configuring Bluetooth through the default App? My understanding is that you should now be able to configure the Bluetooth pairing through that application rather than the exe you mention.
Sincerely,
IoTGirl

mjmeans, 2016-06-04:
Hi. All I have right now that uses Bluetooth LE GATT is a couple of heart rate bands. Both are able to pair as expected using Windows 10 Desktop and use the BLE GATT sample in Windows Desktop. But neither show in Windows 10 IoT in the Bluetooth default app. I'm assuming that the IoT default app is not looking for Bluetooth LE devices since when using the prior command line app you had to specifically switch between normal Bluetooth and Bluetooth LE mode. I'm trying to find a cheap TI Sensor tag so that I can test that specifically, but the model specifically mentioned in the sample is now discontinued by TI. There is a newer model of sensor tag, but I wanted to retest using the exact model specified in the sample. Perhaps your team has one and can verify.

mjmeans, 2016-06-04:
Rather than rely on my laptop's internal BLE adapter to check connectivity to the heart rate devices, I tried the CSR 4.0 adapter I've been using with the RPi and it failed to see any BLE devices but could still see regular BT devices. So I think there is a likelihood that I have a failed CSR 4.0 Bluetooth adapter that no longer reads BLE devices. I've ordered an ORICO BTA-403 device which should arrive on Tuesday. I'll retest and check my results with the latest IoT version at that time.

IoTGirl, 2016-06-06:
Hi Mjmeans,  I am going to close this issue based on your last post. We can re-open or create a new clean request with full repro details if needed.  Sincerely, IoTGirl

closed 2016-06-06 by IoTGirl

mjmeans, 2016-06-06:
BTW, I (partially) disagree that the default App is to be the only way to configure Bluetooth devices. The default App will not necessarily exist on devices being created for turnkey applications. OEM's wanting to create a stand alone IoT device will possibly not want the default App. A better approach in this case would be preserve the Power Shell configuration since it can be automated at the time of initial device configuration. Perhaps this can still be available in the commercial version of IoT. However, I do realize that another approach for turnkey devices/applications could be to integrate the pairing functionality in the default App into the turnkey App, presuming that the license allows that code to be excerpted and used in a third-party commercial product.

IoTGirl, 2016-06-06:
The default app is offered as a sample. The code can be utilized in any application freeing the OEM to offer whatever configuration experience they wish.

mjmeans, 2016-06-08:
Hi IoTGirl,

I have confirmed that the problem with the latest release was indeed a problem with the CSR 4.0 adapter I was using. The ORICO BT-403 does see the BLE devices in both the web interface and in the default app.

IoTGirl, 2016-06-08:
Hi mjmeans,
Thank you for the confirmation! .
 Sincerely,
 IoTGirl