ms-iot/content #882: ReleaseNotesInsiderPreview.md WiFi problem

mjmeans, 2016-05-01:
The file states for version 14322:
"Onboard WiFi drivers are now enabled for the Raspberry Pi 3"
This doesn't appear to work as stated.

On RPI3, Installation of 14322 was done from the web admin "Windows Updates" page while using a RPI2 compatible USB WiFi dongle. After downloading and installation and waiting until it notified for reboot, reboot ad update was done. Confirmed the update version number after it was completed. The onboard WiFI does not appear on the RPI3 connected devices list using HDMI video. The onboard WiFi still does not appear in the Devices web admin page nor does it show on the an HDMI monitor connected to the RPI3.

This RPI3 and NOOBS was purchased directly from Microsoft Store.

ivorb, 2016-05-04:
@alecont could you assign a dev to verify and fix?

IoTGirl, 2016-05-27:
Hi mjmeans & ivorb,  Wifi support was released in 14322 but there were a number of bugs. These should be fixed in 14348 when it is available (These issues were tracked at https://social.msdn.microsoft.com/Forums/en-US/70c2b5ab-9e49-4ec2-961f-c196e85ef5a8/raspberry-pi-3-and-14322-iotuapoobe-stuck-on-welcome-pi3-wifi-issue?forum=WindowsIoT)
Sincerely, IoTGirl

IoTGirl, 2016-06-02:
Fix confirmed

closed 2016-06-02 by IoTGirl