ms-iot/content #922: Please update the release note to version 14366

EdiWang, 2016-06-23:
Recently Windows 10 IoT Core has been updated to 14366. But the release note document is not up to date. 

I notice the I2C ports on 14366 is not working, if there is some workaround please write into the document... It's really blocking me from doing IoT projects...

mjmeans, 2016-06-23:
As stated on the forums, a new image will be up shortly and 14366 has been removed from the update process. 14366 requires a BSP change that is not able to be installed via update. So 14366 must be installed by re-flashing.

IoTGirl, 2016-07-05:
Hi EdiWang,
This issue *only* impacted Raspberry Pi due to the BSP issue mentioned above.  In future, the more information about the exact problem you are hitting with the exact sample you are using would be very helpful.  With the release of the full image with the new BSP we are considering this issue resolved so I am closing this issue.
Sincerest thanks for all of your feedback,
IoTGirl

closed 2016-07-05 by IoTGirl