ms-iot/content #923: Error in BoardComparison.md

mjmeans, 2016-06-23:
The file https://github.com/ms-iot/content/blob/develop/en-US/BoardComparison.md states that audio output is available in analog via 3.5mm jack and digital via HDMI port. However, https://msdn.microsoft.com/library/windows/hardware/dn914597(v=vs.85).aspx states only 3.5mm analog audio.

A quick test with the RPi3 with 10.0.14342.1000 and a Samsung TV as an HDMI monitor did not produce sound output via the HDMI port. I did not check an RPi2. So it does not in fact work at this time. However, in Raspian, digital audio does work through the HDMI port. So while it is a feature of the board, it is apparently, currently, not implemented.

You might want to add a note to the BoardComparison.md file so people will not assume that the digital audio through the HDMI port is something to expect to function at this time.

mjmeans, 2016-07-19:
Since the board specs are on this github repo and the primary purpose of the repo is to support R&D efforts, what I'm suggesting is an (*) next to HDMI audio output and add a footnote that says something like "Only 3.5mm audio output is supported in Windows 10 IoT at this time. Windows 10 IoT may support HDMI audio output in a future build".

That way, any new developers of media server type applications for Windows 10 IoT will not be mislead.

stephenhuang, 2016-09-20:
We have moved board comparison to a new page.

stephenhuang, 2016-09-20:
We've created a bug to address this issue this month. We've created a new page call Board support and SOCs

mjmeans, 2016-09-21:
Would you please provide links to the updated pages?

IoTGirl, 2016-10-31:
Hi MJ,

I believe it is [https://developer.microsoft.com/en-us/windows/iot/explore/deviceoptions](https://developer.microsoft.com/en-us/windows/iot/explore/deviceoptions)

And you get to it by going to WindowsOnDevices.com > Docs> Supported Boards & SOCs.

Sincerely,
IoTGirl

closed 2016-10-31 by IoTGirl