ms-iot/content #933: broken link

dunnleaddress, 2016-07-24:
broken link example:

https://github.com/ms-iot/content/blob/develop/en-US/win10/samples/%7B%7Bsite.baseurl%7D%7D/Resources/images/DigitalSignage/DigitalSign_horses.png

ivorb, 2016-08-01:
Are you viewing the page through github.com UI, or our site?  we know all the links are broken when viewing through github, but ms-iot/content is meant to be viewed through our pages (windowsondevices.com)

IoTGirl, 2016-08-11:
Steps to solve the link issue:
1. Go to WindowsOnDevices.Com
2. Select "Samples" from the top menu
3. On the samples page find Digital Sign (C#)
[https://developer.microsoft.com/en-us/windows/iot/samples/digitalsign](https://developer.microsoft.com/en-us/windows/iot/samples/digitalsign)

This page shows the "Horses" image correctly
Sincerely,
IoTGirl

closed 2016-08-11 by IoTGirl