ms-iot/content #956: Broken Links ?

obastemur, 2016-08-12:
See original discussion at https://github.com/Microsoft/ChakraCore/issues/1428

IoTGirl, 2016-08-12:
Please provide the complete set of steps to see the issue.  Not that the same issue was just closed [https://github.com/ms-iot/content/issues/933](https://github.com/ms-iot/content/issues/933)

IoTGirl, 2016-08-16:
The error on the MSDN issue has been fixed.  Closing.

munyirik, 2016-08-18:
Links are broken again. Try searching for 'lightning' and clicking on the first link presented. You'll get page not found.
Can we please redirect the old links to the new links?

IoTGirl, 2016-08-29:
Please provide the failing links so that they can be traced.

obastemur, 2016-08-29:
>  Try searching for 'lightning' and clicking on the first link presented. You'll get page not found.

> See original discussion at Microsoft/ChakraCore#1428

ivorb, 2016-08-29:
currently testing redirects in staging - should be fixed within 24 hours

ivorb, 2016-09-19:
Redirects have been put in place a little while back, if you are still seeing an issue please open a new issue and assign it directly to me.  Thanks!

closed 2016-09-19 by ivorb