ms-iot/content #961: Remove "cylon wiring" sample

munyirik, 2016-08-18:
I believe this page was created by mistake - https://developer.microsoft.com/en-us/windows/iot/samples/cylonservowiring

It's just the cylon servo sample duplicated.

munyirik, 2016-08-18:
cc @msalehmsft

stephenhuang, 2016-08-22:
Can you clarify add links for the duplicates. This way I can make sure i'm not deleting the wrong file

munyirik, 2016-08-30:
My title is misleading. Here's the problem.
 - https://developer.microsoft.com/en-us/windows/iot/samples/cylonservowiring has nothing to do with wiring. 
- The title says 'servo' but there's no servo in the sample. The instructions are to blink and LED.
- The topic 'Cylon servo controller' in https://developer.microsoft.com/en-us/windows/iot/samples should say something like 'Cylon samples' - the description should say something like 'control a servo and blink and LED with Cylon."

closed 2016-09-30 by munyirik