ms-iot/content #971: Document the IoTStartupTask

TylerMSFT, 2016-08-26:
I reference the IoTStartupTask but we don't have it documented.  Here's some info about it:

It is unique to IoT and not supported in store apps as it essentially runs as a headless service with a very large memory cap. It is part of the IoT package. Below is sample code IoT has put out that uses it:

https://github.com/ms-iot/samples/tree/develop/BlinkyHeadless 

Here is the portion of the manifest referencing it, note the IoT-specific namespace.

<Extension Category="windows.backgroundTasks" EntryPoint="BlinkyHeadlessCS.StartupTask">
          <BackgroundTasks>
            <iot:Task Type="startup" />
          </BackgroundTasks>
        </Extension>

TylerMSFT, 2016-12-02:
Heya,

Not sure what I’m supposed to do with this.

Assigned #971<https://github.com/ms-iot/content/issues/971> to @TylerMSFT<https://github.com/TylerMSFT>.

IoTGirl, 2016-12-02:
Hi Tyler,
We are not sure what to do with this either so I will close it assuming there is not issue here at this point.
Sincerely,
IoTGirl

closed 2016-12-02 by IoTGirl