ms-iot/content #999: Instructions for adding Windows Remote Arduino projects to your solution are slightly wrong

scout208, 2016-10-25:
For the docs at https://developer.microsoft.com/en-us/windows/iot/docs/wra. Under Option 2, Step3, number 3, the location for the Microsoft.Maker.Serial project is slightly wrong given the instructions. The instructions suggest that this project is located in the Microsoft.Maker.win10 folder, but in reality it is located under source/Serial folder in the current state of the `remote-wiring` repository. 

I think either the location of the project should be changed so that it matches the instructions or the instructions should be change, just to avoid confusion.

IoTGirl, 2016-12-02:
Hi Scout,
The screen shots given were based on the steps taken by the author of the document.  
Sincerely,
IoTGirl

closed 2016-12-02 by IoTGirl

scout208, 2016-12-02:
Okay, but I still think it would be more clear to reader and produce less confusion if the screen shots matched the instructions.

turkycat, 2016-12-02:
Connor is correct, actually. The location of the code has moved since those screenshots we're taken. Originally the Serial content was included in the remote-wiring repository, but at one point it was moved to another repository so that layer could be easily included in other projects given that it is the basis for the communication of this library. The screenshots/content were never updated to reflect this.