ms-iot/content #1012: Explain performance

HakanL, 2016-11-15:
I was looking at this page:
https://developer.microsoft.com/en-us/windows/iot/docs/LightningPerformance.htm

and I don't understand why with .NET Native tool chain would be slower (lower frequency) vs without. My understanding is if it's without the native tool chain then it's running MSIL vs with the tool chain it's compiled into machine code. Can you perhaps add some comments to the page explaining if/why this is expected?

IoTGirl, 2016-11-16:
Hi HakanL,

I have reached out to the team and have the following answer for you:

> .Net native has a completely different implementation of managed/native interop than standard .net and it hadn’t been optimized for the very trivial (value type only parameters, single method call on a com interface) interop signatures used in our gpio APIs. The interop transition ended up dominating the cost of everything else and so tanked the throughput. 
> 
> .Net native in general wasn’t done to increase execution speed, it was built to avoid compilation on the client both to improve security (don’t need to be able to execute data pages) and app load time. The perf goal for execution was generally parity with standard .net.

Sincerely,
IoTGirl

closed 2016-11-16 by IoTGirl

HakanL, 2016-11-16:
Thank you, that was very helpful. Do you think it would make sense to add that footnote to the table if others have a similar question?

IoTGirl, 2016-11-16:
Hi HakanL,

You are very welcome!  Based on this thread, the team is reviewing the page as a whole and it will likely be updated with the next OS drop as well.

Sincerely,
IoTGirl