ms-iot/content #1014: PowerShell is there but not there?

goigle, 2016-11-22:


IoTGirl, 2016-11-28:
Hi Goigle,
That link loads for me. Are you sure you are not having connection problems?
Sincerely, IoTGirl

goigle, 2016-11-28:
Sorry, I should have been more clear. That link loads for me, the page it is hosted on says "Page not found."

goigle, 2016-11-28:
OK, so it looks like the page redirects to "page not found" after getting a moved permanently header
[image]

IoTGirl, 2016-11-30:
I am not sure where you are getting that link. It does not go to page not found for me. Have you refreshed your browser cache?

goigle, 2016-11-30:
I got the link from one of your answers on the MSDN forums.
I have checked on 2 different computers using Edge, Chrome, and Pale Moon (Firefox variant) and cleared the caches just to be sure.

Is it possible that your cache is the reason you are not experiencing the problem? What browser are you using?

IoTGirl, 2016-11-30:
I actually use three browsers in my checking - IE Edge & Chrome.  If you send me the link to my answer I will see if that link is bad.

goigle, 2016-11-30:
Ah OK, I see what it is. The link is now a direct one:  
Old: https://ms-iot.github.io/content/en-US/win10/samples/PowerShell.htm  
New: http://developer.microsoft.com/en-us/windows/iot/docs/powershell

The new one does work, thanks!

IoTGirl, 2016-11-30:
Excellent! Thank you for your confirmation.

closed 2016-11-30 by IoTGirl