ms-iot/content #1031: Can you use SSH keys?

benyanke, 2016-12-17:
re: https://github.com/ms-iot/content/blob/develop/en-US/Docs/SSH.md

I see no mention of ssh keys, which are a major method of logging in with SSH.

IoTGirl, 2016-12-22:
Can you provide a description of what you have tried? Do you have a link that explains what you are hoping to accomplish?

benyanke, 2016-12-22:
I'm considering giving this a shot, but I see no way of logging in with SSH keys for Widows IOT, only passwords. Is that possible?

IoTGirl, 2017-01-06:
I am still not clear what you are trying to accomplish. I don't use SSH keys tp log in on my desktop for instance. Can you give and example on how you might do this with other versions of Windows 10?

LordOfDorks, 2017-01-06:
The problem with using SSH with an asymmetric key certificate is that Windows requires the user to provide a password as seeding secret entropy to unlock the account token and get access to the user specific data (CryptProtectData/CryptUnprotectData). This means that when you boot a Windows system and a particular user has not logged on since the start, all his data will remain protected even against a local administrator.

Since SSH only does a challenge response with the private/public key pair, the session manager would not get the password as unique entropy to unlock the account so it would not be able to fully log on the user. Also logging on with an asymmetric key certificate puts the burden of identity and authority management on the IoT device itself and without BitLocker the device cannot protect itself against an offline attack where an adversary would injects public credentials into the image and then would boot the device and just log on.

Having said that I do understand that using cryptographic keys are definitely more secure in protecting the channel, especially if they are held in the TPM. We are currently considering the options what can be done here without jeopardizing the user data protection or system integrity.

IoTGirl, 2017-01-23:
As there has been no response for more than two weeks, I am closing this issue based on Stefan's answer.

closed 2017-01-23 by IoTGirl