ms-iot/content #1034: Any chance of getting the OpenCV sample for C#?

enghch, 2016-12-28:
Are there are any plans to port the OpenCV sample to C#? That would make my life so much easier.

IoTGirl, 2017-01-06:
Hi enghch,

I know of no current plans to convert this sample but I have let the IoT Team know of your request.

Sincerely,
IoTGirl

djaus2, 2017-01-08:
Wrt getting it working in C# possibly some interop might work but then you would lose performance.
Without a native C# version of the code you are stuck with interop.
A native C# version is beyond the cope of things here.
You might be best to push this issue at http://opencv.org/

enghch, 2017-01-08:
Thanks @IoTGirl and @djaus2. I've been working on piecing together an OpenCV-based UWP app (for IoT and non-IoT devices) with a C#/XAML front end. It hasn't been easy to find all the missing pieces but I think I've almost got it all working. As you mentioned, for performance reasons, I've written all the low-level image manipulation in a C++ Windows Runtime Component. All the UI and higher-level functionality is in a C# app that depends on the C++ WRC. I should probably write a blog post on all the steps required to make it work.

closed 2017-01-08 by enghch

djaus2, 2017-01-09:
Yes I agree that is the way to go. UI in C# interoping with the CV code. I would be interested in seeing the code when done and your blog.