ms-iot/content #1039: iot/samples/OpenCV Link error

djaus2, 2017-01-08:
https://developer.microsoft.com/en-us/windows/iot/samples/OpenCV

The GitHub link is missing an h in https at: 

1.	Download and open the OpenCV Solution
Download the working OpenCV library for ARM from **GitHub**. Unzip it to a **plce** you can remember it, and copy the folder’s path (i.e. C:/path/to/opencv/) for the next step.

Ie was ttps://github.com/Microsoft/opencv/tree/vs2015-sample
And plce should be place

PS I couldn't fork this repository to make the correction myself.
The page does invite you to edit it on GitHub

djaus2, 2017-01-08:
It gets worse:
https://github.com/Microsoft/opencv/tree/vs2015-sample
Doesn't exist.
There is no tree folder under 
https://github.com/Microsoft/opencv

djaus2, 2017-01-08:
The issue then is that you need to change the code Branch from Master  to vs2015-samples
or vs2015-samples-ARM:
Change "Download the working OpenCV library for ARM from GitHub" to
**Download the working OpenCV library for ARM or x86 from GitHub:
- Go to https://github.com/Microsoft/opencv
- Change the Branch from master (click on it) to vs2015-Samples-ARM
             OR for x86 to vs2015-Samples**

derekameer, 2017-01-18:
Thanks for the comments on the typos! You can actually access the correct branch directly from https://github.com/Microsoft/opencv/tree/vs2015-samples, which will be in my PR to fix the page. I just forgot to add the "s" on the end.

"for ARM" should be removed, since it works on ARM, x86, and x64. I've corrected this in my PR:
https://github.com/ms-iot/content/pull/1043

closed 2017-01-20 by rachitb777