ms-iot/content #1044: NetworkPacketCapture documentation

cyberh0me, 2017-01-18:
https://github.com/ms-iot/content/blob/develop/en-US/Docs/NetworkPacketCapture.md

following this documentation results in error message

"Unable to start ETW session: MMA-ETW-Livecapture-0a383648-934d-4aba-b89c-d28bca638e52
Host Name: pi3
User ID: Administrator
Remote hosts running windows 8 or earlier do not support remote capture. Otherwise this can be caused by WMI configuration issue, or host name and user credential input error.
 Please configure the WinRM service and add the remote host to the trusted hosts list on the client, by running the following command strings from an elevated command prompt:
   WinRM quickconfig -quiet
   WinRM --% set winrm/config/client @{TrustedHosts="pi3"}

If you are connection to multiple remote hosts, list them with comma separators, as follows:
   WinRM --% set winrm/config/client @{TrustedHosts="remoteHost1,remoteHost2, "}  .,. etc

   Check host name and user credential input, then restart your live trace."

i recommend you add that powershell part ...
[image]

IoTGirl, 2017-01-23:
Hi Jordan,
Can you please review?
Sincerest thanks,
IoTGirl

cyberh0me, 2017-02-08:
done
br
Andre

IoTGirl, 2017-02-08:
Merging based on Jordan's review. Thanks for the pull, Andre!

closed 2017-02-08 by IoTGirl