ms-iot/content #1055: OpenCV UWP C++ Example

baqwas, 2017-02-01:
Using the instructions for the article on **OpenCV - UWP C++ project**, I was able to Build _All_ with no failed or skipped messages for x64 version. Yet when I start Debug in x64 mode, I get the following error messages:

- 'HelloOpenCV.exe' (Win32): Loaded 'C:\Windows\System32\ntdll.dll'. Cannot find or open the PDB file.
- 'HelloOpenCV.exe' (Win32): Loaded 'C:\Windows\System32\kernel32.dll'. Cannot find or open the PDB file.
- 'HelloOpenCV.exe' (Win32): Loaded 'C:\Windows\System32\KernelBase.dll'. Cannot find or open the PDB file.
- 'HelloOpenCV.exe' (Win32): Loaded 'C:\Windows\System32\combase.dll'. Cannot find or open the PDB file.
- 'HelloOpenCV.exe' (Win32): Loaded 'C:\Windows\System32\ucrtbase.dll'. Cannot find or open the PDB file.
- 'HelloOpenCV.exe' (Win32): Loaded 'C:\Windows\System32\rpcrt4.dll'. Cannot find or open the PDB file.
- 'HelloOpenCV.exe' (Win32): Loaded 'C:\Windows\System32\bcryptprimitives.dll'. Cannot find or open the PDB file.
- 'HelloOpenCV.exe' (Win32): Loaded 'D:\Users\Reza\Documents\Nextcloud\Visual Studio 2015\Projects\HelloOpenCV\x64\Debug\HelloOpenCV\AppX\opencv_core300.dll'. Symbols loaded.
- 'HelloOpenCV.exe' (Win32): Loaded 'D:\Users\Reza\Documents\Nextcloud\Visual Studio 2015\Projects\HelloOpenCV\x64\Debug\HelloOpenCV\AppX\opencv_imgproc300.dll'. Symbols loaded.
- 'HelloOpenCV.exe' (Win32): Loaded 'D:\Users\Reza\Documents\Nextcloud\Visual Studio 2015\Projects\HelloOpenCV\x64\Debug\HelloOpenCV\AppX\opencv_imgcodecs300.dll'. Symbols loaded.
- The thread 0x1154 has exited with code -1073741515 (0xc0000135).
- The thread 0x3834 has exited with code -1073741515 (0xc0000135).
- The program '[11956] HelloOpenCV.exe' has exited with code -1073741515 (0xc0000135) 'A dependent dll was not found'.

How do I resolve the issue? Thanks.

IoTGirl, 2017-02-01:
Hi baqwas, 
The "PDB" files are just debug database files for component. You should be able to ignore those (See [https://www.codeproject.com/questions/1076518/how-to-solve-c-windows-system-ntdll-dll-cannot-fin](https://www.codeproject.com/questions/1076518/how-to-solve-c-windows-system-ntdll-dll-cannot-fin) ).

The issue you want to review is "The program '[11956] HelloOpenCV.exe' has exited with code -1073741515 (0xc0000135) 'A dependent dll was not found'" as it tells you that a DLL it is expecting is not available.  I would recommend you redo the steps around "https://developer.microsoft.com/en-us/windows/iot/samples/opencv  Add dlls to your project" to confirm that all DLLs are actually available.

Sincerely,
IoTGirl

IoTGirl, 2017-02-01:
PS: Please also review the closed issues at [https://github.com/ms-iot/content/issues?q=is%3Aissue+is%3Aclosed](https://github.com/ms-iot/content/issues?q=is%3Aissue+is%3Aclosed)

closed 2017-02-01 by IoTGirl

baqwas, 2017-02-01:
Appreciate the quick response. Sadly your recommendations will not fix the problem. My apologies for jumping the gun twice! (I guess I get disqualified to enter issues?). The PDB issue was a non-issue. I simply wasn't thinking straight because everything had gone smoothly until then. The DLL issue is a red herring. The app launches but a dialog message pops up stating "Unable to activate Windows Store app"

I have searched the web for a work around for this error (which is the one that is preventing the app from fully launching) and none of those workarounds apply in my situation.

How does one activate a Windows Store app when I am simply building a test UWP app? Thanks.

P.S. I will work on a clean machine with a fresh VS 2015 install and OpenCV library build.

IoTGirl, 2017-02-21:
Hi Baqwas,
Please research your new error, Unable to activate Windows Store app, directly.  I did a web search for "Unable to activate Windows Store app" and found quite a few posts.
Sincerely,
IoTGirl

baqwas, 2017-02-22:
Thanks for the update. Will review in due course. Presently working on sensors with VS 2017 RC and not encountering any issue with my trivial examples.

 

Regards.