ms-iot/content #1062: File Transfer Protocol documentation

cyberh0me, 2017-02-09:
[https://github.com/ms-iot/content/blob/develop/en-US/Docs/FTP.md](url)

"Find IoTStartupOnBoot.cmd, right-click it, and click Edit."
the Owner of this File is TrustedInstaller not Administrator(s) so Edit results in Access denied

an information to change the owner should be added to the documentation with a link to a page that shows how this can be done

as it seems no such page exists for Windows 10 on a public available MS server (at least i did not find one...) a new page "Take ownership of files" should be added with screenshots...

br
Andre

IoTGirl, 2017-02-09:
Hi Andre,
If there is not a document in Windows On Devices where you feel this should be added, I recommend going to the Windows 10 docs on MSDN and giving feedback there as the recommendations given there get given to the Documentation team at Microsoft.

Otherwise, please submit a pull against the page where you feel this should be added and I will get it reviewed.

Sincerely,
IoTGirl

IoTGirl, 2017-02-09:
Note: I assume this issue is as a result of the MSDN post [https://social.msdn.microsoft.com/Forums/en-US/03f037f9-d621-4058-99e2-3c47393a203c/raspberry-win10-how-to-start-automatically-ftpd?forum=WindowsIoT](https://social.msdn.microsoft.com/Forums/en-US/03f037f9-d621-4058-99e2-3c47393a203c/raspberry-win10-how-to-start-automatically-ftpd?forum=)WindowsIoT

cyberh0me, 2017-02-09:
your assumption is correct

IoTGirl, 2017-02-11:
Thanks Andre!!! [https://github.com/ms-iot/content/pull/1067](https://github.com/ms-iot/content/pull/1067)

closed 2017-02-11 by IoTGirl