ms-iot/content #1065: Wrongly swapped link for C++ and C# projects

maroonedsia, 2017-02-10:
Hi,

Apparently some of links for C++ and C# projects are wrongly swapped. For example, currently, under "New Samples" we have 

https://developer.microsoft.com/en-us/windows/iot/samples/alljoyndsbcs for C++, and
https://developer.microsoft.com/en-us/windows/iot/samples/alljoyndsb for C#

But, it should be the other way. There might be more similar issues.

IoTGirl, 2017-02-11:
Hi Maroonedsia,
Is the problem just the name with "Cs" being used for the C++ sample or is there a link somewhere that is failing?
Sincere thanks,
IoTGirl

maroonedsia, 2017-02-13:
Hi IoT Girl! :D

No, the links are actually wired wrong. If you go here, you can see: https://developer.microsoft.com/en-us/windows/iot/samples

C++ link goes to C# project, C# link goes to C++ project. 

Cheers,
Sia

IoTGirl, 2017-02-22:
Hi Sia,
This should be fixed based on the merge above. 
Sincerely,
IoTGirl

closed 2017-02-22 by IoTGirl