ms-iot/content #1098: while (true) - for Listen() doesn't work for cancelling

ramiss, 2017-03-17:
Hello!

First, thank you for this, it has been very helpful. I wanted to point out that the when cancelling the token the while (true) continues to run and lock up the app.  I was able to solve it by replacing while (true) with:
> while (!ReadCancellationTokenSource.IsCancellationRequested)

Hope that helps!

Richard

IoTGirl, 2017-03-17:
Hi Richard,
Are you pointing to a specific sample? Can you give more detail as to the setup of your issue?
Sincere thanks,
IoTGirl

ramiss, 2017-03-17:
Hello!

My apologies!  I am referring to the Listen() function for the following:
https://github.com/ms-iot/content/blob/ba4fd011423e2325b72b6417d6459d9607c8f5af/en-US/Samples/SerialUART.md

Richard

jordanrh1, 2017-03-20:
ReadAsync() should through an OperationCanceledException when the cancellation token is canceled. Is an exception not being thrown?

ramiss, 2017-03-20:
It was throwing that exception, but once I caught the exception I could not escape the loop.  I need to be able to disconnect and reconnect multiple times.

closed 2017-12-05 by saraclay