ms-iot/content #1099: Any plans to support I2S bus?

shawaj, 2017-03-19:
Just wondered if there were any plans to support the I2S bus at all?

ooeygui, 2017-03-20:
Hi @shawaj,
I2S can be used by an audio driver to communicate with off SOC components. We do not have plans to add a UWP API for I2S. 

What are you looking to enable?

shawaj, 2017-03-21:
We have a range of high resolution audio boards for Raspberry Pi (and soon
also Intel Joule) under the name JustBoom which communicate over I2S

We would love to see the ability to use this in IoT core.

IoTGirl, 2017-03-24:
Hi @shawaj.
Can you give us a little more detail as to your requirements for I2S for your addon board? 
-	DMIC and speaker configuration
-	Which codec are you planning to use
-	Which features are you planning to enable?

Sincere thanks,
IoTGirl

shawaj, 2017-03-27:
Hi @IoTGirl,

Speaker configuration for DAC and Amps 2.0
Speaker configuration for Digital outputs 2.0, but can do digital
passthrough of 2.0, 2.1, 5.1, 7.1 or pretty much anything else if the
source supports it (i.e. passthrough of Dolby digital 5.1 to be decoded
further down the chain)

Codec is DAC PCM5122
Amp is compatible with DAC codec
Digital board uses WM8804 codec

Features: Digital volume control, analog gain.

Hope that helps?

@ooeygui - you can email my GitHub handle @gmail.com :-)

shawaj, 2017-04-04:
@ooeygui @IoTGirl - just checking you saw this?

IoTGirl, 2017-04-04:
Yes!  Seen and forwarded.

shawaj, 2017-04-04:
Awesome :-)

IoTGirl, 2017-04-07:
Hi @shawaj,
Just to set expectations, based on the responses so far, I don't think there will be a fast answer to I2S support.  My understanding is that a number of options are being evaluated by Intel for the Joule.  They will publish the details when they are available.
Sincerely,
IoTGirl

shawaj, 2017-04-07:
Ok cool, well if anything changes on that please let us know :-)

rachitb777, 2017-04-07:
Closing the issue.

closed 2017-04-07 by rachitb777

shawaj, 2017-04-07:
Doesn't seem like it was resolved yet?

IoTGirl, 2017-04-12:
Hi @shawaj ,

This answer is pending a response from Intel and they will post to their forums.  

Sincerely,
IoTGirl

shawaj, 2017-04-14:
So does that mean Microsoft won't bring I2S support to the RasPi in IoT core?

IoTGirl, 2017-04-21:
That is correct. Microsoft won't but the OEM potentially could.  In this instance Intel is investigating I2S support for the Joule.

shawaj, 2017-04-21:
So would that support then be ported to the RasPi too?

If we wanted to build I2S support for Pi with IoT core, what would that
involve?

Thanks

IoTGirl, 2017-04-21:
My assumption is you would need to work with the I2S manufacturer to create a driver for your scenario.  If you type "i2s driver" in your favorite search engine you will see the hardware folks that offer those drivers.  You would want to reach out to them and see if they are making an IoT Core driver for ARM platforms like the Pi.

shawaj, 2017-04-22:
The hardware driver for the DAC chip is not the issue here. We can easily
sort that.

The issue is the support for the I2S bus within IoT core that that driver
would need to interface with. That seems like something that needs to be
done by the IoT core devs? Would be happy to pitch in and help wherever
possible but want to know the best process for progressing this so we don't
waste time (we are only a very small team) on stuff that isn't useful or
helpful.

Thanks

IoTGirl, 2017-05-05:
Hi shawaj,
The answer from April 4 was a reference to the Windows side. There is a universal driver. Please do follow up with the I2S manufacturer for their support of the standard interface.
Sincerely,
IoTGirl

shawaj, 2017-05-05:
Sorry, I am not following exactly that you mean here? Are you saying I2S is already supported in IoT core?

IoTGirl, 2017-05-05:
Here is the answer again, given on April 4th above:

> Their board has a different codec (WM8804 – Cirrus Logic typical does not support WinIoT universal driver very well) and it seems that it is for playback only (one speaker, no dmics).

The guidance is to work with the OEM

shawaj, 2017-05-05:
When you say work with the OEM you mean the chip manufacturer? Texas
instruments?

Again, what I am after is some documentation or a datasheet about creating
a driver for our codec - pcm5122 - or any codec for that matter.

Sorry if that wasn't clear. Not meaning to be obtuse just not really sure
what you're not understanding.

Thanks

IoTGirl, 2017-05-06:
Cirrus Logic is not your OEM?

shawaj, 2017-05-06:
No, we make our own boards for the RasPi. We use the Texas Instruments
PCM5122 DAC and an amp which is compatible with same driver. We then use
the Cirrus Logic WM8804 but that's a digital S/PDIF output.

But not sure what you mean by "OEM" in this scenario as we make the
hardware ourselves. Do you mean the chip manufacturer?

Thanks

IoTGirl, 2017-05-16:
Yes.

shawaj, 2017-05-16:
Pcm5122 is Texas instruments
WM8804 is cirrus logic

How can we support them?

IoTGirl, 2017-05-17:
I don't know what further answer to give you beyond restating the points already given.

> Their board has a different codec (**WM8804 – Cirrus Logic typical does not support WinIoT universal driver very well**) and it seems that it is for playback only (one speaker, no dmics).

shawaj, 2017-05-18:
@IoTGirl not sure if I am being clear.

*We* want to create the driver - how do we do this? Can you provide me with the documentation / process for us to develop our own?

What is "the universal driver"?

Thanks