ms-iot/content #1100: Adding support for sensors/HATs etc?

shawaj, 2017-03-19:
What's the current process for adding support for HATs, sensors and add ons in IoT core?

Excuse me if I've just completely missed the documentation for this but have not been able to find anything.

Thanks very much

IoTGirl, 2017-03-20:
Hi @shawaj ,

There are number of sensor and hat based projects you can leverage to learn about Sensors/Hat on IoT Core.  I highly recommend you take a look at hackster.io for projects like [https://www.hackster.io/laserbrain/windows-iot-sense-hat-10cac2](https://www.hackster.io/laserbrain/windows-iot-sense-hat-10cac2) 

As to adding a sensor the question really comes down to the type of sensor you are trying to add and how it communicates.  For example there are multiple accelerometer samples at [https://developer.microsoft.com/en-us/windows/iot/samples](https://developer.microsoft.com/en-us/windows/iot/samples)

I recommend you look at the [community page](https://developer.microsoft.com/en-us/windows/iot/Community) for more places to search as well. 

Sincerely,
IoTGirl

closed 2017-03-20 by IoTGirl

shawaj, 2017-03-21:
> laserbrain/windows-iot-sense-hat-10cac2
> com/en-us/windows/iot/samples

The reason I asked is because we have a few HATs and add ons we would love
to add support for such us an epaper display, audio boards etc

Soon for the Intel Joule too :-)

Just wondered what the process to add support looked like?

IoTGirl, 2017-04-07:
Hi @shawaj,
Did you review the samples mentioned in my previous answer?
Sincerely.
IoTGirl

shawaj, 2017-04-07:
The boards we want to interface are an epaper display which communicates over SPI, and a temperature sensor and real time clock that communicate over I2C.

Then there's the I2S audio cards too and mobile data boards which communicate over SPI and I2C.

As I said, what's the process for building in support?

IoTGirl, 2017-04-13:
Hi @shawaj,
The process to build support for any hardware is to create a driver.  There is a lab for creating a driver for IoT Core here [https://developer.microsoft.com/en-us/windows/iot/samples/driverlab](https://developer.microsoft.com/en-us/windows/iot/samples/driverlab).
Once that driver is created you can add that driver into your BSB for your custom OS build. To find out more about creating your own OS image please see [https://msdn.microsoft.com/en-us/library/windows/hardware/dn916104](https://msdn.microsoft.com/en-us/library/windows/hardware/dn916104%28v=vs.85%29.aspx)
Sincerely,
IoTGirl