ms-iot/content #1109: LCD_Rotate=2 for Pi3 Touch Screen overwritten during Windows Update

JT73, 2017-03-26:
I have a bunch of production devices using the 7" Touch Screen in the Case for the Pi3. 
In the Config.txt file LCD_Rotate=2 rotates the screen so that it's up the right way.
During Windows Update this setting is getting overwritten and causing devices to display upside down.

IoTGirl, 2017-03-27:
Hi @JT73 ,
Have you built your own custom image for the Pi or are you using the Microsoft image?  The image provided by Microsoft is intended for testing only and will push OS updates as well as BSP updates.  It is a BSP update that will change the config.txt. 
Sincerely,
IoTGirl

JT73, 2017-03-27:
Hey IoTGirl? This Justine? 

Yes, we are deploying using the IoT Dashboard as this is a set of lab/demo devices that take people through building the device, configuring and adding an app so that they have a demo device for showcasing Windows IoT and Azure IoT. 

Is there a way to protect ourselves from getting these updates ?

thanks
Jay

IoTGirl, 2017-03-27:
Hi Jay,
Yes! To create your own image and not get the BSP updates for your platform you need to build the image like an OEM would.  The instructions are available here: [https://msdn.microsoft.com/en-us/windows/hardware/commercialize/manufacture/iot/iot-core-deployment-and-imaging](https://msdn.microsoft.com/en-us/windows/hardware/commercialize/manufacture/iot/iot-core-deployment-and-imaging) and the specific section is **Configuring IoT Core update settings**

> To configure an image to receive only OS updates, you must configure generic POP packages. 

Sincerely,
IoTGirl

IoTGirl, 2017-04-04:
Hi Jay,
I have escalated this question to the team but the answer is this is currently "By Design" as that file may require changes in the event of a BSP update.  It is part of the RPi BSP so not immune to update by the OEM of the image.   The only way to control BSP update is to be the OEM yourself.

Sincerely,
IoTGirl

closed 2017-04-04 by IoTGirl