ms-iot/content #1128: Display a date on the Whats New page

enghch, 2017-04-16:
I would love to see a "release date" on the What's New page, or even just a date that the page was last modified. It's always confusing when you do google/bing search which leads you to a page and you have no idea if it was written last month or last year.

IoTGirl, 2017-04-17:
Hi enghch,
The page [https://developer.microsoft.com/en-us/windows/iot/Docs/WhatsNew](https://developer.microsoft.com/en-us/windows/iot/Docs/WhatsNew) is version release driven rather than date driven.  The mention of Creators Update is expected to identify the version details.

For interim or Insider updates, these are offered through the Insider program itself.

Sincerely,
IoTGirl

enghch, 2017-04-17:
OK, I can accept that. Still, if your team decided to update that page during the creator's update timeframe (let's say two months from now) we wouldn't have an indicator that something had changed on the page since the last time we read it. But I suppose if I suspected the page looked different I could always click through to the GitHub repo and see the actual change history.

IoTGirl, 2017-04-17:
Hi enghch,
That is correct.  The change history in GITHUB will reflect the latest update as well offer you the ability to see exactly what those changes were.  The assumption is that the contents should reflect what we know to be true for "What's New" at the time of release of that milestone.
Sincerely,
IoTGirl

closed 2017-04-17 by enghch