ms-iot/content #1137: Rectanges not appearing over faces Emotion demo

HolyBreadstick, 2017-04-24:
https://www.microsoft.com/cognitive-services/en-us/emotion-api This guide has rectangles appearing over faces. I followed it and I can see the code triggering to load the rectangles however they never appear over faces.

HolyBreadstick, 2017-04-24:
Solution, change where it gets the size of the ImageCanvas. double resizeFactorH = ImageCanvas.ActualHeight / decoder.PixelHeight;
                double resizeFactorW = ImageCanvas.ActualWidth / decoder.PixelWidth; Need actual width not just width.

IoTGirl, 2017-04-24:
Hi @HolyBreadstick,
That link refers to their forum at https://stackoverflow.com/questions/tagged/microsoft-cognitive and not this GITHUB.  https://github.com/ms-iot/content/blob/f8143637f263a6c92b3cdd123b5b5afc9cbfd59e/en-US/Docs/ComputerVision.md 
If you feel there is a change required to the code in this GitHub, please submit a change request for review.
Sincerely,
IoTGirl

closed 2017-04-24 by IoTGirl