ms-iot/content #1142: Use powershell command to setup and start service

georgeslegros, 2017-04-26:
I could not get the 2 following lines to work on my raspberry pi 3

```bash
sc config SEMgrSvc start=auto
sc start SEMgrSvc
```

I thing it would be better to encourage people to use Powershell to do this and use the following commands

```powershell
Set-Service SEMgrSvc -StartupType "Automatic"
Start-Service SEMgrSvc
```

It worked like a charm for me.

rachitb777, 2017-04-27:
We've added instructions for both in the sample page.

closed 2017-04-27 by rachitb777

rzdhop, 2023-04-13:
It's been some time since this post, but I'd like to add additional information.
_(I've seen this issue bc I stepped upon almost the same problem)_

When using the command :
```powershell
sc config SEMgrSvc start=auto
sc start SEMgrSvc
```
He was using PowerShell command `Set-Service` & `Start-Services`
But the syntax provided is for `sc.exe` which is different from `sc` !

When using `sc.exe SEMgrSvc start= auto` it should work !
Or use `sc SEMgrSvc -StartupType "Automatic"`
```
C:\Users\%USER%> Get-Alias sc

CommandType     Name                                               Version    Source
-----------     ----                                               -------    ------
Alias           sc -> Set-Content
```