ms-iot/content #1175: Cortana not enabled on Windows 10 IOT (OS Version: 10.0.15063.251), Raspberry PI 3 Model B

brvinodh, 2017-07-04:
Hi, 

I have updated to the latest build of Windows IOT and have done the following as per the documentation. 

1. Plugged in a Microphone with Speaker, this is recognized on the Device Portal Page as well, i have attached the screenshot. 
[image]

2. Raspberry Pi 3 is connected to Monitor via HDMI cable

3. Im not able to turn on Cortana; the radio button is disabled, The screenshot is attached below. 

[image]

4. Option "Start Cortana on Boot" is also not available in the Device portal home Page. 

Can you please help.

Thanks
Vinodh

rachitb777, 2017-07-13:
@brvinodh Can you check if Cortana app is installed on the image. You can do this by running 
`iotstartup list` using PowerShell or SSH.

zhuridartem, 2017-08-22:
Hi brvinodh.
This is a symptom of Cortana App not present in the image. 
An update does not add new optional feature packages, and Cortana App is included in such package. 
Easiest way to pick it up is to clean install latest image.

brvinodh, 2017-08-23:
Hi zhuridartem, 

Thank Let me perform a fresh install and check back.

closed 2017-08-23 by brvinodh