ms-iot/content #1180: Installing and Servicing apps on Windows 10 IoT Core > Special Instructions for Headless Apps

AlexanderPilhar, 2017-07-14:
I have problems creating the neccessary appxupload-package for an Background Task project following the docs: [Store.md](https://github.com/ms-iot/content/blob/develop/en-US/Docs/Store.md)

Details are on StackOverflow: [VS2017 fails to create appxupload-package of headless IoT app for store publishing](https://stackoverflow.com/questions/45099438/vs2017-fails-to-create-appxupload-package-of-headless-iot-app-for-store-publishi)

I wonder if the Special Instructions for Headless Apps are correct!?

saraclay, 2017-11-29:
Hi @AlexanderPilhar, thank you for reaching out. I've assigned this issue to a few folks from our team to look into.

paulmon, 2017-11-29:
According to the StackOverflow link this issue has been resolved.

closed 2017-11-29 by saraclay