ms-iot/content #1188: support for the pi camera

karann-msft, 2017-07-20:
I was speaking to a customer who was doing something very interesting with Win 10 IOT and electric cars. He wants to be able to control the pi camera that connect to the board using the CSI camera port.
Is this supported? Either yes or no, is this documented?

cyberh0me, 2017-07-28:
for future questions not related to hosted github projects please use https://developer.microsoft.com/en-us/windows/iot/Community
the release notes you will find on the download page https://developer.microsoft.com/en-us/windows/iot/Downloads contain information about Raspberry Pi Camera Support

saraclay, 2017-10-12:
@karann-msft Hi Karan, it would be in your customer's best interest to reach out to Raspberry Pi to express the desire to use the Pi Camera using the CSI camera port.

closed 2017-10-12 by saraclay