ms-iot/content #1190: Windows 10 IoT Preview Build 16237 - Raspberry Pi Password Not Working

Tornadocraver, 2017-07-23:
Using the IoT Dashboard app, I flashed a (more than qualified) SD card with Insider Preview Build 16237. Now, when I turn on the Raspberry Pi 3 B and try to access it from the Device Portal (or any other connection), it says that the password is incorrect. I tried re-flashing the SD card, thinking that I typed/pasted the wrong password, but that did not fix it. I submitted a problem in the Windows Feedback app a few days ago about this, and it has still not beecommented on or fixeded. My RPi has been on for a few days now (I just left it alone), because the SD card has a high chance of getting corrupted when I do not shut down before pulling the power (shutdown also requires the password). What's weird is that, a few days prior to me flashing with build 16237, I had flashed another Pi with 16241: a build that at least works with the password. Now, 16237 is still the only build that I can get (I've tried everything I could think of. Please let me know the status of this problem if you're able to, because the rest of my project is on hold until this is solved.

cyberh0me, 2017-07-28:
please dont abuse github for not github related issues, please use the community for developer questions or the feedback hub if you discover any issue

saraclay, 2017-10-12:
@Tornadocraver Hi Micah - Did you use p@ssw0rd? That is the default password to use before you go into the Device Portal.

Tornadocraver, 2017-10-12:
@saraclay Hi Sara,

I did try using "p@ssw0rd" as the password (and all other combinations with the "Administrator" username), but to no avail. There seemed to be an issue with the SD card that I was using in this particular Pi though it was brand new. I finally resorted to buying another SD after exhausting all my other options, and decided to flash with the stable build to be safe. Everything worked after that.

Thank you!

closed 2017-10-12 by Tornadocraver