ms-iot/content #1191: Windows 10 IoT Preview Build 16241 - IoT Remote Client Shows Blank Screen

Tornadocraver, 2017-07-23:
Apart from my other issue on the other thread, it seems like Preview Build 16241 has an issue with the IoT Remote Client app. The Raspberry Pi 3 B is running Windows 10 IoT Insider Preview Build 16241. I did check (multiple times) that the remote server is enabled in the Device Portal. When I first try to connect, the little loading circle spins for a few seconds and then disappears (I assume that means that it's connected).

[image]

However, after the loading circle disappears, all I get is a white screen.

[image]

I went to the Device Portal and took a few screenshots which appear normal. I also plugged the Pi into my TV via an HDMI cable, and that works fine as well. Any ideas on what is going on here? This project is on hold for me (in addition to the other one) until it gets resolved. Thanks for your time!

cyberh0me, 2017-07-28:
please dont abuse github for not github related issues, please use the community for developer questions or the feedback hub if you discover any issue

saraclay, 2017-10-12:
@Tornadocraver Hi Micah - Is this issue still persisting for you?

Tornadocraver, 2017-10-12:
@saraclay Hi Sara,

I am sorry I didn't post that I resolved this issue a while back (as I procured the solution on another forum). I ended up flashing the latest stable build on the Pi that was giving me the issue. By the time I wanted to try the preview build again, a new version had been released and it no longer gave me the issue.

Thanks for reminding me about this, and I'll close it now.

closed 2017-10-12 by Tornadocraver

ErikParso, 2020-05-03:
> @saraclay Hi Sara,
>
> I am sorry I didn't post that I resolved this issue a while back (as I procured the solution on another forum). I ended up flashing the latest stable build on the Pi that was [...]

Hello, in what version is this fixed ?