ms-iot/content #1197: Good page, needs more screenshots

1RedOne, 2017-07-27:
This is a super informational post.  I would like it to be ammended a tiny bit and include some info:

* If I wanted to integrate an SPI display in addition to an attached HDMI display, is there any specific changes I need to make?
 * Best practices for working with a display like this.  For instance, we generally don't worry about update time or rates or things like that when wokring with Windows IoT.  Windows handles the updates.  Since this is closer to embedded programming, are there pitfalls we can avoid?
 * Easiest one: Screen shots!  You can't give a cool walkthrough like this without a demo of the screen working.  I'm pretty sure that violates the Geneva Convention.

saraclay, 2017-10-12:
@1RedOne Hi Stephen - could you elaborate on which this page you're referring to? It does not show up in issues. Thanks!

1RedOne, 2017-11-01:
HI Sara, sorry for the radio silence.  I was referring to this page here,

https://github.com/ms-iot/content/blob/8c7bb023d45e9a8ff6d4ea91e5566466a17503f9/en-US/Samples/SPIDisplay.md

saraclay, 2017-11-29:
Got it. We have given our samples a bit of a new look, but I don't think that has answered your questions. I will assign this to someone to look into :)

SlavyMihov, 2017-12-06:
* You can control an SPI peripheral (for example a display) without any concerns for affecting the HDMI display. Both SPI and HDMI are completely separate interfaces.
 * [Best practices working with an SPI display](https://www.bing.com/search?q=Best+practices+working+with+an+SPI+display) is a huge topic. There are plenty of details available depending on what you need (SPI interface best practices, serial displays best practices, embedded devices with display(s) best practices, etc.).
 * What demo of the screen working would make the [SPI display](https://github.com/Microsoft/Windows-iotcore-samples/blob/master/Samples/SPIDisplay/CS/README.md) sample more informative? Would a picture of the display showing "Hello world" make the sample easier to discover, understand and reproduce?

1RedOne, 2017-12-07:
Wow, HelloWorld sounds great! I've since solved my issues but I think
content like this would have really helped me do it faster.

closed 2018-01-17 by saraclay